olliw42/storm32bgc #7: MOST usful feature suggestion

Tommixoft, 2017-01-10:
Hi, yesterday i though of super feature to add to board - current draw info. SO people can see in "real time" how much current is drawn so by this we can know how good we balanced gimbal, If you dont have ability to add amps - you maybe can think something by using info you sending to motor controllers. SO if there is high amount of correction sent -this means bad calibration (of course if gimbal is in steady position).
I think such feature would be great.

Tommixoft, 2017-01-12:
sad this is dead project.

noradtux, 2017-01-12:
Try to contact olliw on rcgroups.com. This project is far from dead, but it's not using github as it's primary development plattform.

Tommixoft, 2017-01-12:
@noradtux RC groups is filled with noob questions and in 1 hour my suggestion would be 5 pages behind :) So i put here. What do you think about my feature? Would it be useful to you?

noradtux, 2017-01-12:
@Tommixoft I think using the NT-logger provides better means of evaluating how well the system is balanced. So no, I wouldn't consider measuring current draw to have any benefit over what is already there.

Tommixoft, 2017-01-12:
NT logger works with NON nt firmware? Cause all that nt stuff is overrated. i in my entire using never got even single I2C error.

noradtux, 2017-01-12:
Well, development of non-NT-stuff has effectively halted. NT isn't just about better rejection of interference, it also gives benefits like the nt-logger and camera modules, giving me more functionality with less wires. I could even reduce wires more if I used the motor modules.

Tommixoft, 2017-01-12:
Hmmm camera modules? what is this? Maybe i will buy NT sensor some day, my country makes them, anyways, im using this board as gimbal for camera, not on drone, so so far it provides all needed functions, but i wish firmware be open source, now everything depends on 1 person.

noradtux, 2017-01-12:
I use a cc3d atom as camera-IMU. I can use two of its servo outputs to control the camera. I think you will find the wiki enlightening.

olliw42, 2020-01-08:
the hardware can't do current measuerment
outdated

closed 2020-01-08 by olliw42