olliw42/storm32bgc #15: IMU not available

Vishranthee, 2020-01-08:
I tried my level best and went through all the RC group pages and re-flashed the firmware multiple times. But IMU is not available and I'm unable to calibrate it by any means.

Power LED is solid.  @olliw42 

[image]

olliw42, 2020-01-08:
you probably have a I2C-based IMU and not an NT IMU, ie., use the wrong firmware. Please read up in the wiki: http://www.olliw.eu/storm32bgc-wiki/Downloads

the primary location for questions & discussions is btw here: https://www.rcgroups.com/forums/showthread.php?2055844-STorM32-BGC-32-bit-3-axis-brushless-gimbal-controller

olliw42, 2020-03-26:
no further response so seems to be resolved and I close :)

closed 2020-03-26 by olliw42