olliw42/storm32bgc #17: Firmware source code

serghov, 2020-04-29:
Hi,

I couldn't find much information about the firmware, is it open source?

elihunter173, 2020-05-04:
The firmware is not open source. From the website

> The o323BGC firmware is free (but not open source). Besides unlimited private use you are also granted the permission to use it for commercial purposes under the condition that (1) you don’t modify the firmware, e.g. remove or change copyright statements, (2) provide it for free, i.e. don’t charge any explicit or implicit fees to your customers, and (3) correctly and clearly cite the origin of the firmware and the project web page in any product documentation or web page.

olliw42, 2020-05-23:
elihunter173 thankfully answered
closing

closed 2020-05-23 by olliw42