olliw42/storm32bgc #19: 10 magnets on motor at 0.90 i2c

whoim2, 2020-09-30:
Hello! Have motors with 10 magnets, but in program 0.90 may set minimum 12. Motor work, but have hissing sound in many positions.
What im may set 10 magnets?

whoim2, 2020-10-01:
im try to set 10 poles via set command in "rc input", command 20, 26, 32. Value 10, and store to eeprom. Read button and log write: 10 poles. But strange hissing sound present.

olliw42, 2020-10-03:
use a latest firmware, it supports 10 poles
it needs to be however a NT setup, and not I2C

whoim2, 2020-10-03:
> use a latest firmware, it supports 10 poles
> it needs to be however a NT setup, and not I2C

Not have nt imu module (
 0.96 fw not support 10 poles, if im set value from setcommand?
Ok, i understand. Thanks.

closed 2020-10-03 by whoim2

olliw42, 2020-10-03:
I think the approach you've tried should have worked, not sure why it didn't
maybe you best bet is to patch the GUI exe. It's perl, and Perl is an interpreter, so in the exe you just find the source packed with the interpreter. Hence, it could be possible to just patsh the limits for the motor poles.
Have fun.

whoim2, 2020-10-03:
> I think the approach you've tried should have worked, not sure why it didn't
> maybe you best bet is to patch the GUI exe. It's perl, and Perl is an interpreter, so in the exe you just find the source [...]

Maybe, hissing sound not because of the poles?
Im try today verify poles at log and tests

whoim2, 2020-10-03:
> I think the approach you've tried should have worked, not sure why it didn't
> maybe you best bet is to patch the GUI exe. It's perl, and Perl is an interpreter, so in the exe you just find the source [...]

You may say about compile software for perl? pl file found at 0.96 folder. Latest active perl for win?
Have win7 and ubuntu