olliw42/storm32bgc #24: unable to calibrate "imu1 is not present or on healty!"

onlineapps-cloud, 2023-09-14:
hi, i tried to calibrate my gimbal, i have installed imu2, and he working as expected, but when i want to calibrate i cannot bypasss this erro, in wizard i disabled imu1 check, but not help :(
can anyone say, what i do wrong?
[image]
[image]
[image]
[image]

onlineapps-cloud, 2023-09-14:
in wiki http://www.olliw.eu/storm32bgc-wiki/Using_a_2nd_IMU
The calibration of the 2nd IMU works similar to the camera IMU; one just has to select the 2nd IMU in the respective fields. 
but, this not work, i had errol that "imu1 is not present or on healty!"

onlineapps-cloud, 2023-09-14:
[image]

onlineapps-cloud, 2023-09-14:
[image]

onlineapps-cloud, 2023-09-14:
sorry, im newbie, i have only gimbal, i need additional camera with IMU? can you give some example?
[image]

onlineapps-cloud, 2023-09-14:
as i understand this sensor not workin, yes? IMU2 is accelerometer that is onboard, yes?

olliw42, 2023-09-14:
pl go first to the wiki, it will answer nearly all your questions ... you are expected to have spend sufficient time with the wiki, otherwise you will not get to anywhere with STorM32.

Read espeically also big red warning boxes, like this one here: http://www.olliw.eu/storm32bgc-wiki/Downloads

closed 2023-09-14 by olliw42

onlineapps-cloud, 2023-09-14:
i searched my IMU module, but i cannot find it here http://www.olliw.eu/storm32bgc-wiki/Overview_of_NT_Modules
my IMU module is not listed,
module close view. Model: INVENSENSE MPU-6050
[image]

Ensys NT Modules uses MPU-6050, which firmware i need to load, to work this IMU module?
[image]

onlineapps-cloud, 2023-09-14:
Also i want donate (if you accept donations) for future development (and for your great work!), i can pay in crypto only.

onlineapps-cloud, 2023-09-14:
[image]
now i understand why it will not work, i need firmware that will support this IMU device, yes? if yes, i don't know which firmware support this IMU module :(, i still reading  documentation :)

onlineapps-cloud, 2023-09-14:
yes, you are correct, firmware work as expected, my apologies!