olliw42/storm32bgc #28: Gerber Files for NT Modules (IMU)

natiz, 2024-03-24:
Hi @olliw42,

Thanks for this project! Glad to see its still active.

I have an old 1.31 board which I want to revive (been sitting in storage since 2015), and am looking for an NT IMU module for it, as I only have I2C IMUs, which are limited to the v0.96 firmware.
I saw the very limited options for [sourcing the NT modules](https://www.olliw.eu/storm32bgc-wiki/Where_to_get_NT_Modules), so I thought about maybe sourcing the board myself, via one of the PCB manufactures (such as JLPCB), which require Gerber files.
Would you be able to export those in Gerber format, for example for the [storm32-nt-imu-v42-g4](https://github.com/olliw42/storm32bgc/tree/master/storm32-nt)?

olliw42, 2024-03-24:
I haven't done that for a very long while and would have to relearn and work out what to do and what is needed essentially form scratch. I'm sorry to say that I won't find the time to do so in the near future.
I would think that with some web research there should be tools which allow you doing this. Eagle CAD is after all quite popular. 
Sorry.

natiz, 2024-03-24:
Thanks for the quick response.

I’ll try to export it myself, should be possible according to [this](https://jlcpcb.com/help/article/9-How-to-Generate-Gerber-and-Drill-Files-in-Autodesk-Eagle).

Are there any other places that sell some version of the IMU?

olliw42, 2024-03-24:
ensys sometimes has some in stock

closed 2024-03-26 by natiz

olliw42, 2024-03-26:
many THX
you coud consider doing a PR to add them in a subfolder storm32-nt/storm32-nt-logger-v2x-g4/gerber

natiz, 2024-03-26:
Sure https://github.com/olliw42/storm32bgc/pull/29