olliw42/storm32bgc #31: Which firmware should be used for burning the Storm32-nt-bgc-v41 board?

suiaiyun, 2024-08-07:
Hello, may I ask which firmware should be burned onto the Storm32-nt-bgc-v41 board? The corresponding board was not found in the GUI tool. Does this board already have firmware support?

olliw42, 2024-08-08:
this board is fully supported but it is not in a current release yet. Somewhere in the rcgroups thread there is a prerelease version posted (but I wouldn't know where to find it exactly). I'll plan to do a release next week if possible, it when should be in the GUI.

suiaiyun, 2024-08-13:
> this board is fully supported but it is not in a current release yet. Somewhere in the rcgroups thread there is a prerelease version posted (but I wouldn't know where to find it exactly). I'll plan to do a release next week if possible, it when should be in the GUI.

Great, looking forward to this release. Thank you.

sksrmsl, 2024-11-12:
I have two "storm32-nt-bgc-v41" boards.
I'm checking the operation.
The latest version of "o323BGCTool_v269e" does not contain "storm32-nt-bgc-v41" version of the HEX file.

Is there a way to check it out?
I would like to ask for your help

OrhanKoseoglu, 2024-11-22:
> this board is fully supported but it is not in a current release yet. Somewhere in the rcgroups thread there is a prerelease version posted (but I wouldn't know where to find it exactly). I'll plan to do a release next week if possible, it when should be in the GUI.

I can't find appropriate firmware for the v4.1 board?

OrhanKoseoglu, 2024-11-22:
> Great, looking forward to this release. Thank you.

Did you find?

suiaiyun, 2024-11-23:
> Did you find?

No, not yet. I'm waiting for Ollie to release it

kami4ka, 2025-01-27:
Is it dead?

olliw42, 2025-05-02:
v4.1 firmware is included in latest release v2.70e

so closing