olliw42/storm32bgc #35: BGC v90 mit Arducopter nutzbar?

Rothammel, 2025-03-24:
habe einen alten Gimbal, kann man den nutzen? ich habe mav1 eingerichtet und kann ihn im MP als zweite Komponente auswählen und dort steuern, RC Input gelingt mir aber nicht

olliw42, 2025-03-25:
es gibt kein "BGC v90" im STorM32 Ecosystem. Die Vermutung liegt also nahe das es sich um kein STorM32 Gimbal sondern etwas Anderes handelt und du hier mit deinen Fragen am falschen Ort bist.
Zudem sind github issues nicht für allgemeine Supportfragen gedacht. Dafür gibt es entsprechende Threads etc. pp auf rcgroups. Ohne mehr/bessere Infos um welches Gimbal du hast wirst du aber auch dort keine besseren Antqqorten bekommen.

Da es sich um kein Issue sondern Support handelt, schliesse ich das hier.

closed 2025-03-25 by olliw42

Rothammel, 2025-03-25:
danke für deine Antwort. es ist ein Board V1.3 mit Firmware 0.90. ich bekomme keine serielle Verbindung zum laufen, nur Mavlink. Mavlink selber läuft aber auch nicht gut und funktioniert auch nicht mit den Ardupilot Gimbal Funktionen.

ich habe noch etliche dieser Gimbals und auch Aliexpress verkauft sie noch immer.
wo kann ich die V0.96 Quellen ansehen?

olliw42, 2025-03-25:
> Board V1.3 mit Firmware 0.90

ok ... dir ist klar dass das uralt ist (auch v0.96 ist uralt), und daher (schon laaaannnggeeee) nicht mehr supported wird. Sorry man, time goes on.

>  ich bekomme keine serielle Verbindung zum laufen, nur Mavlink. Mavlink selber läuft aber auch nicht gut und funktioniert auch nicht mit den Ardupilot Gimbal Funktionen. 

da kann ich nichts zu sagen, kann alles mögliche sein. Musste dir selber deinen Weg mit hilfe des Wikis und der ArduPilot docs suchen, oder jemand finden der das benutzt.

> auch Aliexpress verkauft sie noch immer.

was nicht heist dass das aktuell wäre ...

> wo kann ich die V0.96 Quellen ansehen?

nirgends

Rothammel, 2025-03-25:
Vielen Dank für die Infos. Leider kann ich aktuelle Boards mit aktueller Firmen nirgendwo kaufen. Ich habe noch etliche dieser gimbles und sie funktionieren einwandfrei an einem copter und auch an Flugzeugen. ich würde sie schon gerne verwenden